dcpedit/pillzmod #12: Register Shifter connected on low frequency pin in case of Nice!Nano

dhruvinsh, 2024-11-12:
Register Shifter is connecto to the pin 13, 14 and 16. but in case of nice!nano those are low frequency pin. ZMK recommands to use high frequency pins which are highlighted in dark blue color.

[image]
[image]

HeavyDutySoul, 2024-11-12:
Anybody assemble on n!n? It work?

asmodeus812, 2024-12-28:
> Anybody assemble on n!n? It work?

@HeavyDutySoul have been using the nice nano v2 with the pro pcb version for a few weeks, seems to be working just fine,

dcpedit, 2025-01-16:
@dhruvinsh , thanks for pointing this out.  I chose those pins because they matched up with the SPI pins for the Pro Micro.  Do you know if using the low-freq pins have any adverse effects on the shift register functionality?  It seems to be working fine so far in my build.

dcpedit, 2025-03-24:
Thanks for finding this info. I’ll need to create a separate Nice!Nano-compatible PCB so I can reroute the traces to use the high-frequency pins. I can’t modify the existing PCB without breaking compatibility with the Pro Micro, whose SPI pins can’t be moved.

cub-uanic, 2025-10-18:
@dcpedit any info when n!n-compatible pcb could be created?
Thanks!

dcpedit, 2025-10-18:
I was purposely stalling on this since I felt like the Pillbug build was the superior one.  But now that it's not available (again), I'll revisit this next chance I get.

dcpedit, 2026-02-14:
@nol00p, awesome work!  Thanks for doing this.  Great idea with the ESC key which will fix issue #24