dcpedit/pillzmod #17: Optional through-hole diodes on thumb PCB (feature request)

rwaldura, 2024-12-27:
Hello! Great project, super useful to bring back to life some older keyboards — my 1999 "Classic" thanks you!
I really appreciate how you provided the option to use either SMD or through-hole components for the main PCB, e.g. the LED resistors.
Would it be possible to achieve the same, with the thumb PCB? It appears only SMD is supported at this time. 
I'd like to have the option to wire through-hold diodes, rather than SMD diodes, on the thumb PCB.
Thanks!

dcpedit, 2025-01-16:
This will not be difficult since the "left hand" diodes are wired differently than the "right hand" diodes.  In order to do this, I will need to use two diode footprints per key position, or simply just create 2 separate PCBs (one for left and another for right).

rwaldura, 2025-01-16:
I see. This had not occurred to me.
Thank you for looking into it!

closed 2025-03-24 by dcpedit

rwaldura, 2025-03-24:
Sorry, the resolution here is that this feature will not be supported? Just want to know, thanks!