dcpedit/pillzmod #19: Triggering key and next in column on first rows

cedounet, 2025-02-02:
Hello,

I built the pro-version and used https://keeb.supply/products/0xcb-helios as a controller which is pin compatible to the pro... Everything works fine (on what I tested at least) but the first two rows.  

I am using rows/col as per the schematics... so opposite of what it is physically on the keyboard 

- Pressing one key on this row will systematically trigger the key _and_ the one below (next column)
   - _= _toggles _=_ and _tab_, tab toggle _tab_ and _capslock_
   - _1_ toggles _1_ and _q_ etc
   - It does that for those 3 first rows

- Same for the end key on thumb cluster that will also trigger a c after it (I reused the original thumb cluster)

I first thought this would be a soldering issue, but:
- I am having a hard time to see where a short would create this... would be all keys i.e. 1,q,a,z, not just the next one
- I have 2 keyboards... 2 controllers and both have the same issue on same rows...
- Using tweezers to activate the switch works flawlessly no triggering of next key. 

Base are contour keyboards (PS2, and AT...) if that is relevant and I had some issues with them but not that one as far as i can remember and definitely not that systematic.

Any ideas on what could cause it or how I could debug that? 

Thanks!

C

cedounet, 2025-02-05:
Looks like I need to use external pull down resistors… O had understood the RP2040 had internal ones but maybe they are not strong enough.  I only tried quickly but if confirmed I will close that issue… 

Thanks

cedounet, 2025-02-22:
Works with pull down…

closed 2025-02-22 by cedounet

dcpedit, 2025-02-23:
This is interesting because I tested this with an Elite Pi which also uses an rp2040 and the internal pull downs were sufficient. Maybe I'll make a note of this in the readme.

cedounet, 2025-02-23:
Most key works… just those 2/3 rows.  As I said earlier I somewhat dismiss soldering issues as it happens on 4 combination of PcB and left key wells.  

I could clearly see the 2 line going down briefly on the scope etc… and have half baked theories about why… but an extra pulldown sorted it.

I wanted to add the Helios or the readme but not sure if that is the reason it behaves differently from Elite-pi?